trustcrypto/OnlyKey-Firmware #172: LED is absurdly bright even at the lowest setting

RE-Specto, 2024-05-19:
hopefully can be addressed at firmware level. regular Onlykey (not the duo model) the indicator LED at its lowest setting (1 = dimmest) is still absurdly bright to the point that I see it in the bedroom at the second floor with the laptop at the table on the first floor, and the glow of the LED reflected by the stairs.

grangerx, 2024-11-20:
With the OnlyKey in the black protective silicone case, it is possible to slip a piece of paper, or possibly silvered anti-static bag, between the OnlyKey and the case, such that the light is diffused/lessened by passing through the material.

It's not as nice as a real firmware fix, but at least it lowers the eye-blistering without having to use the OnlyKey app.

RE-Specto, 2024-11-26:
> without having to use the OnlyKey app.

thing is, I have used the onlykey app to set brightness to the minimum, yet its still ridiculously bright (well it was even worse prior to changing the setting in the app). while a band aid solution can help, I have opened this ticket because I believe there is a bug with regards to brightness calculation, other users may be affected (it is possible that my specific onlykey have a hardware flaw causing incorrect led voltage or something, but I highly doubt that).

I suspect that a large portion of the users are using their onlykey on a laptop laying flat down, where LED is facing down and less visible. in my case, its on a side usb port on my monitor, and LED is facing away from me towards the room.