trustcrypto/OnlyKey-Firmware #174: Passkey Support

freddyyeddy, 2024-09-19:
Now it seems google, and other major companies are moving towards the fido passkey model (https://fidoalliance.org/passkeys/). Would we be able to get this supported with the onlykeys (DUO and or OG).

onlykey, 2024-09-20:
"Passkey technology is the cybersecurity industry’s attempt to unify, streamline, modernize and rebrand existing authentication lexicon, even if the underlying technology is essentially identical to FIDO2/WebAuthn, which has existed since 2018."

OnlyKey supports passkeys, this is essentially the same as passwordless FIDO2

ElonGaties, 2024-11-20:
could issues like these be marked as resolved? as the question/issue seems to have been answered, i dont see much reason in leaving the issue open

closed 2024-11-20 by onlykey