trustcrypto/OnlyKey-Firmware #175: Unlocking OnlyKey PIN using terminal

mihakrumpestar, 2024-10-16:
After reading all the docs I can't find any way to unlock OnlyKey PIN from the terminal. This apparently works for YubiKeys.

For example:

- Tested: `pamu2fcfg` just waits if OnlyKey is locked, Yubikey apparently prompts for PIN
- Untested: unlocking LUKS encrypted partitions not possible without unlocking the key physically (eg. with onboard buttons), Yubikey apparently prompts for PIN

The OnlyKey CLI also does not have any way to unlock the key (to enter the PIN).

onlykey, 2024-10-16:
OnlyKey PIN is entered on the device itself and cannot be done from the terminal.

onlykey, 2024-10-16:
The OnlyKey app is only capable of entering PIN for the OnlyKey DUO. This tradeoff is because the device keypad on the DUO is small. The full size OnlyKey requires entering PIN on device, this way there is no chance of software capturing a user's PIN.

mihakrumpestar, 2024-10-16:
Sorry for not being clear upfront, I am talking about the OnlyKey DUO.

onlykey, 2024-10-17:
It would be possible to do with the OnlyKey CLI, we may be able to get this in the next release, if you would like to give it a try we do accept PRs.

mihakrumpestar, 2024-10-17:
Thank you for the clarification. If I find the time, will give it a try.

closed 2024-10-17 by mihakrumpestar